ixjf/MSIRGB #131: Driver load failed, prob related to Riots Vanguard?

wughuf, 2020-07-18:
Hello there, 
i wanted to try this out with my b450 tomahawk max cause msi's dragoncentre sucks but i ran into the following issue (screenshot down there). At the same time when i tried to install it riots vanguard gave me a notification that it would have to be disabled or sth, dont remember it exactly (screenshot down there too). I didnt try yet to disable vanguard and install it and then enable vanguard again but thought this might be interestin so im putting it up here. 

[image]
[image]

wughuf, 2020-07-18:
okay i just saw there is already a thread on this and no solution yet, sorry for opening another one

ixjf, 2020-07-18:
No problem.

closed 2020-07-18 by ixjf