ixjf/MSIRGB #132: MSIRGB not working anymore after Windows update 2004

Darferth, 2020-07-19:
As the title states, it shows an error message and won't open the tool. Already disabled Riot Games anti-cheat tool "Vanguard". Before the update it was fully working. I can provide logs if someone shows me how to.

ixjf, 2020-07-19:
What error message?

Darferth, 2020-07-19:
Failed to load driver. This could be either due to some program interfering with msirgb driver or a bug. Please report this on.... Basically it was doind that bc of Vanguard but even if I disable the auto start and restart the pc it pops out

ixjf, 2020-07-20:
I can't test whether this happens with me right now. Ideally, you should reboot your computer in diagnostic mode (disable all non-Microsoft services and startup programs in msconfig) and try again to rule out some program interfering with MSIRGB.

Darferth, 2020-07-20:
I confirm that it's not a program that interferes but a bug, in diagnostic mode (safe mode) it gives the same error.

Darferth, 2020-07-20:
I think that it is caused by the new Windows major update, 2004. Before this update there weren't problems at all. Maybe some security patches that deny the tool from executing properly.

Darferth, 2020-07-20:
Forgot to say that I'm currently on Tomahawk B350 plus, idk if that matters. It worked previously

PrzemyslawSegrecki, 2020-08-03:
same thing with B450M Mortar MAX :(
[image]

ixjf, 2020-08-04:
I can't reproduce. Works on my machine on Windows 10 update 2004. Will have to investigate further. Give me some time.

Darferth, 2020-08-04:
Can we help you providing any logs? Will try again with a fresh installation

ixjf, 2020-08-04:
If you can, that would be nice. If that doesn't work, I'll make a debug build that logs what's going on so I can try to figure it out.

PrzemyslawSegrecki, 2020-08-06:
ok, my problem is solved: D
1. I tried uninstalling and installing MysticLight - no results
2. Then i deleted Mystic Light
3. The next step was to install Dragon Center
4. He didn't work either (No settings for MysticLight)
5. But after updating motherboard BIOS Dragon Center started working and MSIRGB too :)

ixjf, 2020-08-06:
Your issue, however, @PrzemyslawSegrecki, seems to be different from what the OP described, so the OP likely shouldn't bother following your solution.

phamalex20, 2020-08-15:
@ixjf I can confirm the same issue happens to me on Windows version 2004 as well.

ixjf, 2020-08-15:
Huh, okay then. I'll make a debug build soon and ask that you provide me the logs afterwards.

ixjf, 2020-08-15:
Try this:

[MSIRGB-v2.2.1.2-debug.zip](https://github.com/ixjf/MSIRGB/files/5078847/MSIRGB-v2.2.1.2-debug.zip)

Run it and then post the contents of the file debug_log.txt created, if you will.

phamalex20, 2020-08-17:
[image]
The debugger is returning this error, sorry for the late response.

phamalex20, 2020-08-17:
Nevermind, I just needed to click okay. Here is the debug_log.txt 
[debug_log.txt](https://github.com/ixjf/MSIRGB/files/5086670/debug_log.txt)

ixjf, 2020-08-17:
What debugger are you referring to? Anyway, this seems to be some weird Windows 10 issue. Can you check if installing the latest WIndows 10 updates and/or disabling the antivirus fixes the issue?

Darferth, 2020-08-17:
The debug build actually works, WU made some updates and now ligths load and work as behaviored. Probaly is just as @ixjf said.

phamalex20, 2020-08-17:
I'll check updates, I don't have an antivirus of windows firewall enabled either.

phamalex20, 2020-08-17:
I'm on latest build of windows with all updates installed. It used to work for me but I guess there's some kind of software conflict.

ixjf, 2020-08-18:
> [image]
> The debugger is returning this error, sorry for the late response.

When does this happen?

phamalex20, 2020-08-18:
Oh thats from a debugger @ixjf sent me. It works, it just runs that dialogue after I open it.

ixjf, 2020-08-18:
That certainly shouldn't be appearing, though. Do you only get that message with this debug build, or did this message show up with release versions of MSIRGB?

phamalex20, 2020-08-18:
Only the debug build.

ixjf, 2020-08-18:
Maybe some Windows update that hasn't been rolled out to everyone? Darferth reported that after some Windows updates, it started working.

Darferth, 2020-08-18:
I will provide the list of updates I have installed later today! They came up as available on August 15 for me

Scarica Outlook per Android<https://aka.ms/ghei36>

ixjf, 2020-09-05:
Did you fix this?

phamalex20, 2020-09-06:
Unfortunately, it doesn't work anymore. It's mystifying why. It worked
about two months ago I know for sure.

ixjf, 2020-09-06:
Does release 2.3.0 still not work?

phamalex20, 2020-09-06:
Oh, I had not realized there was an update. Let me test it when I get home
later tonight. Thank you for letting me know.

high1, 2020-09-14:
Having the same issue - MSIRGB was working fine.  Windows 10 Version 2004 19041.508.
[debug_log.txt](https://github.com/ixjf/MSIRGB/files/5220052/debug_log.txt)

ixjf, 2020-09-14:
Can't reproduce on 19041.450. Might be again Windows 10 updates fucking something up. @phamalex20, did the update fix the issue?

phamalex20, 2020-09-14:
No, the error output is still the same. Windows is strange in its ways.

high1, 2020-09-15:
The issue in my case was Riot Vanguard, as you asked in https://github.com/ixjf/MSIRGB/issues/148.

phamalex20, 2020-09-15:
> The issue in my case was Riot Vanguard, as you suggested in #148

Oh! Let me check that

ixjf, 2020-09-15:
That could be the issue. I had ruled it out in this case as @Darferth said he had disabled Vanguard. I would ask all of you to completely uninstall it and try again, then report back.

high1, 2020-09-15:
It's easy to test - exiting the Vanguard by right clicking the tray icon resolves the issue here. There is a notification that one driver is blocked on Vanguard startup, though...

sv1k, 2020-11-04:
I have the same issue
I updated BIOS on my b450 Tomahawk MAX and LED just turned on
So I'm on Win 2004 19041.572
Never had Vanguard or any Riot games
Tried 2.2.1.2 version and it's not working. 
Tried 2.3.0 as well. Not working either
Debug error 1450
start_drv: StartService failed with error 1450
load_drv: start_drv() != ERROR_SUCCESS
load_drv failed

ixjf, 2020-11-04:
Does it start if you run Windows in diagnostic mode? Win key + R -> type 'msconfig' -> tick 'Diagnostic mode' -> apply and reboot. Run msconfig again and tick normal boot to go back to normal.

sv1k, 2020-11-05:
In Diagnostic mode it doesn't launch at all
Nothing happen after trying to launch app

ixjf, 2020-11-05:
Right, diagnostic mode might not work. Instead, change to 'normal startup', then go to the 'Services' tab, tick 'Hide all Microsoft services', click 'Disable all', apply, OK. Then go to the 'Startup' tab and do the same. Disable all startup items. Reboot. See if MSIRGB works. 

To go back to normal, do the same thing and enable the services/startup items again. Make sure to make a list if you're not sure which ones were enabled and which weren't.

sv1k, 2020-11-05:
Nope :(
Doesn't work, got the same error 'Failed to load driver'
Any extra ideas ?

ixjf, 2020-11-06:
Did this start happening after a Windows update like the OP explained, or did this always occur for you?

sv1k, 2020-11-09:
Can't say for sure
Last time I disabled lights about 2 month ago
Maybe it's a windows update. Maybe BIOS update (currently v37 tomahawk max)
Maybe some software
But for the past year your program worked like a charm.. Until now
If you have any ideas I can make additional investigation

ixjf, 2020-11-11:
I'm honestly clueless. I'm trying to think of why this might be happening, but so far nothing's come up. Whenever I have the time, I'll make another debug build to try to get to the root of the issue.

marcelomartinsIT, 2020-12-02:
Well, same problem here, failed to load driver ;) 2004 update installed...

Me-Phew1, 2020-12-06:
Exactly the same problem
B450 Tomahawk
Win 2004
I had Vanguard but I uninstalled it a long time ago
Debug error 1450
start_drv: StartService failed with error 1450
load_drv: start_drv() != ERROR_SUCCESS
load_drv failed

ixjf, 2020-12-06:
I'll try to make time this week to figure this out.

Me-Phew1, 2020-12-06:
Never mind, I finally fixed it by uninstalling Faceit AC. I forgot I even installed it. However the debug build still says my motherboard isn't supported while the normal build doesn't but I guess it's just a minor issue.

ixjf, 2020-12-06:
The debug build is older than the latest version of MSIRGB, which adds support for some motherboards, hence why.

Can everyone else confirm that the issue is caused by Faceit AC? Or any other anti-cheat software.

sv1k, 2020-12-07:
Yeah. Can confirm it's caused by Faceit AC. Even if just installed and not running at the moment

phamalex20, 2020-12-07:
That's... Strange, I'll try to reproduce that on my own PC later today.

phamalex20, 2020-12-07:
I can confirm this worked!!! Also, Vanguard dislike msirgb so I can't play Valorant at the same time.. Uninstalling Faceit AC worked to make this work.

ixjf, 2020-12-07:
Mystery solved then. Anti-cheats all block MSIRGB. Should put a notice on the main page, if nothing else can be done about it.

marcelomartinsIT, 2020-12-11:
I also have the faceit AC installed and haven't tried to uninstall it for testing purposes.
Although most of the people has confirmed it has a problem with it.

DethMagnetic, 2021-01-22:
Yep, had the exact same issue with op, disabling Vanguard fixed it on my B450M Mortar MAX. Vanguard seems to really be fucking shit up with its kernel level access.

closed 2021-02-04 by ixjf