ixjf/MSIRGB #133: B450 Tomahawk max not supported

JulianTheMen, 2020-08-04:
When I open the program it says that my motherboard is not supported altough it says in the README it is supported. I would like to know if the b450 tomahawk max is supported or not.

ixjf, 2020-08-04:
Can you run this command on a command prompt and show me the result?

`wmic path Win32_BaseBoard get Manufacturer, Version, Product`

JulianTheMen, 2020-08-04:
> Can you run this command on a command prompt and show me the result?
> 
> `wmic path Win32_BaseBoard get Manufacturer, Version, Product`

I will try that tomorrow

JulianTheMen, 2020-08-05:
[image]

Here you go

ixjf, 2020-08-05:
When you start MSIRGB, it'll prompt you whether to continue or not despite not recognizing it as a supported motherboard. Your motherboard is _reported_ to work, I.e. other users have tried it and told me it works fine. The reason the program doesn't tell you everything is okay is because it is not a motherboard that *I* guaranteed MSIRGB would be compatible with. I'll have to update MSIRGB at some point to include more motherboards.

In short, you should be fine.

closed 2020-08-05 by ixjf