ixjf/MSIRGB #135: Archive of custom scripts

dmsergeevich63, 2020-08-08:
I think I didn’t write something brilliant, but I don’t understand why the repository does not yet have a theme with all custom scripts? Or what did I not see? Yes, there are three standard scripts, is that really all?
My script runs in 3 modes by the hour:
7 to 21:
Slow color change and full brightness
21 to 22: the same, only with minimum brightness
22 to 7: minimum brightness and red, it has the lowest brightness compared to other colors and is quite beautiful
My computer is quite close to the sofa and at night or while watching movies it can stand out a lot, I have to adjust the backlight manually!
```
Lighting.SetStepDuration(0)
Lighting.SetFlashingSpeed(0)
Lighting.SetBreathingModeEnabled(false)
local i = 0
local t
local saturation = 0.933
local brightness = 1
local delay = 800 -- delay between each colour update, in milliseconds
local colour_step = 1.1

--depends on your cooling, marks start and end of color wheel
local Tmax=83
local Tmin=40

while true do

		
		print (os.date ("%H"))
		t=tonumber(os.date("%H"))
	if t<=21 and  t>=7 then
		print("день")
		if t==21 then
		saturation = 0.001
		else saturation =0.933
		end
    local r, g, b = Lighting.ColourUtils.HSVtoRGB((i % 98.0) / 98.0, saturation, brightness)

    r = tonumber(("%x"):format(r * 15), 16)
    g = tonumber(("%x"):format(g * 15), 16)
    b = tonumber(("%x"):format(b * 15), 16)

    Lighting.BatchBegin()
    for i = 1, 8 do
        Lighting.SetColour(i, r, g, b)
    end
    Lighting.BatchEnd()

    os.sleep(delay)

    i = i + colour_step
	
	else
	print("ночь")
	
		local r, g, b = Lighting.ColourUtils.HSVtoRGB((1.666-1)%1, 0.001, brightness)
		r =tonumber(("%x"):format(14), 16)
		g =tonumber(("%x"):format(15), 16)
		b =tonumber(("%x"):format(15), 16)
		
		Lighting.BatchBegin()
		for i = 1, 8 do
			Lighting.SetColour(i, r, g, b)
		end
		Lighting.BatchEnd()
		os.sleep(delay)
	end
end
```

ixjf, 2020-08-16:
There are 4 scripts in the repository, which are the ones that I created to demonstrate potential usage of MSIRGB. Anyone is free to contribute and add different effects. 

In your case, it seems to be the same effect as the Hue Wheel but tailored to your specific use case, so I don't know if it should qualify as a different effect.

Also, while I appreciate the effort and you sharing your work, it does not meet my code quality standards, so I wouldn't add it to the repository as-is (bad indentation, repeated code, unused variables, useless prints, and code that I don't really understand, specifically, the part where you define `r`, `g`, `b` for night time).

ixjf, 2020-09-06:
If you would like to review your code for submission, I will leave this issue open. Otherwise, I'll close it soon.

closed 2021-02-04 by ixjf