ixjf/MSIRGB #136: MSI B450M Bazooka MAX WiFi - Disabled LED, toggles Bluetooth!?

Gryd3, 2020-08-16:
Please see the title.
On first run, the 'Mystic Light' device disappears from the 'Devices' Section of Windows Settings.

When attempting to run the 'Hue Wheel.lua' script, the windows USB Add and Removal sounds play on a loop.
During this process, an 'Intel(R) Wireless Bluetooth(R)' device intermittently shows up in the device list.

If you time it just right, you can manually hit the 'Stop Running Script' and the new Bluetooth device will persist.
Alternatively, you can edit the 'Hue Wheel.lua' script.
Edit Line 16 : "while true do"  to  "while i < 45 do"

DXDiag System Information partial  : 
         Operating System: Windows 10 Enterprise N 64-bit (10.0, Build 19041) (19041.vb_release.191206-1406)
                 Language: English (Regional Setting: English)
      System Manufacturer: Micro-Star International Co., Ltd.
             System Model: MS-7C87
                     BIOS: 1.00 (type: UEFI)
                Processor: AMD Ryzen 5 3600X 6-Core Processor              (12 CPUs), ~3.8GHz
                   Memory: 32768MB RAM
      Available OS Memory: 32718MB RAM
                Page File: 5728MB used, 32108MB available
              Windows Dir: C:\Windows
          DirectX Version: DirectX 12
      DX Setup Parameters: Not found
         User DPI Setting: 96 DPI (100 percent)
       System DPI Setting: 96 DPI (100 percent)
          DWM DPI Scaling: Disabled
                 Miracast: Available, with HDCP
Microsoft Graphics Hybrid: Not Supported
 DirectX Database Version: 1.0.8
           DxDiag Version: 10.00.19041.0084 64bit Unicode

Bluetooth Device instance path:
USB\VID_8087&PID_0AA7\6&3AF0F9CE&0&9

Mystic Light Device missing from Device Manager.  (Switched off at hardware level?)

Gryd3, 2020-08-16:
Additional information : 
MSIRGB-v2.2.1.2
Recovery of the 'Mystic Light' device can be done by clearing CMOS.

ixjf, 2020-08-16:
Glad to know you were able to fix the problem. Other than that, it seems clear that your motherboard is not compatible with MSIRGB. At this moment, I'm not planning on adding support for other motherboards, since it's difficult to test MSIRGB without access to one of them.

closed 2020-09-05 by ixjf