ixjf/MSIRGB #138: RGB not Working

thexihad, 2020-08-26:
I have msi b450 tomahawk max.After using This app . MSI mystic Light is not working.The only color I can see is Red Or I can disable all effects To turn off lights.

My Mobo is bricked?? 
How can I fix it to use MSI mystic Light again?
Please help guys.

ixjf, 2020-08-26:
Shutdown your computer. Turn off the power supply. Press the power button for about 10 seconds. Turn the PC on again. Should get everything working again. Alternatively, remove the CMOS battery for a few minutes, then put it back on.

thexihad, 2020-08-26:
Okay I'm doing it right now

thexihad, 2020-08-26:
After doing that ..Msi Dragon center is loading.. Initialization.
[image]

thexihad, 2020-08-26:
No bro it's not working .. If I choose any effect in Mystic light ..Only red is showing ,and then turning off sometimes. No other Colors. What else can I do?? 

When I used you App first time it showed other colors too.. But then I choose disable all effects.. I don't know after that..The only color shown is red.

ixjf, 2020-08-26:
You probably didn't press the power button for enough time. Either try repeating the process or use the alternative process - remove the CMOS battery for about 10 minutes, then put it back on.

thexihad, 2020-08-27:
I removed the cmos battery for 3-4 mins. I'll try again Today and let you know .I hope it works.

ixjf, 2020-08-30:
Did you fix it?

thexihad, 2020-08-30:
No bro it's not fixed. I gave up I don't know what to do. And there's no new bios update available. Otherwise I would have tried updating BIOS. 
Is there anything else that I can do?

ixjf, 2020-08-30:
I don't think you're doing the process right. I think it's very unlikely MSIRGB could do any irreversible damage on the motherboard. Could you repeat the first process just to make sure? And be patient. Power supply OFF. Press the power button on the case for *at least* 10 seconds. Then turn everything on again.

If this still doesn't work, I'll try to see what I can do. But it should be fixable.

thexihad, 2020-08-30:
Okay bro I will do the process again Tomorrow . Pressing power button and Removing cmos battery for 10 min also... I will let you know... And Did anyone use MSIRGB with B450 Tomahawk max and had no problem?? . First time the colors were changing with MSIRGB..But after that it's Red Only... I'll do the processes And let you know again.

ixjf, 2020-08-30:
The problem you describe has been reported to happen on that motherboard. MSIRGB won't work properly with it _at this time_. However, no one has complained that they couldn't get lights to work properly again.

thexihad, 2020-08-31:
[image]
[image]
[image]
[image]
[image]

Look Brother . Removed battery for more than 10 mins. 
Now I'm using A lite version windows 10 . It's missing powershell so I can't check with Mystic light .So I want to use MSIRGB but you just look at the ui Does it look all right? Cause there's no Script. 
Or can You send Me the proper MSIRGB with scripts I'll download it from Gdrive if you can.

ixjf, 2020-08-31:
Did you download the scripts?

thexihad, 2020-08-31:
Yes but I don't know how to use them.😰

ixjf, 2020-08-31:
Did you follow the instructions on the main page on how to setup MSIRGB?

thexihad, 2020-08-31:
Update #
Color is Changing using scripts ..But colors looks
Very fake I mean not very good.. I'll send you video. And fan speed is varying while Changing colors (This can Harm my fan) slowing down while Changing colors

thexihad, 2020-08-31:
[image]
I turned of other effects because of the changes in fan speed. 
And I can't use Any SINGLE COLOR option.
Those 8 slots are black and when I customize it only show white.

And in my Case fan I see multiple color LED thats why I think color not looking good.Never saw that before.

ixjf, 2020-08-31:
Well, I'm glad the motherboard is working properly.

Now, as I said, MSIRGB does not support your motherboard at the moment. I'll try to add support ASAP, but can't promise anything.

thexihad, 2020-08-31:
I hope you'll add it . Will be waiting 🌹

closed 2020-08-31 by thexihad