ixjf/MSIRGB #139: Explicit Support for B450 Tomahawk MAX

FluorescentGreen5, 2020-08-31:
Since the B450 Tomahawk MAX has been recommended a lot for people who are building a system, it would be a good idea to have explicit support added for it. Right now, it will sometimes work (with inverted colours), cause the motherboard to cycle between colours even when you have every square the same colour or even do nothing at all.

ixjf, 2020-08-31:
I'll work on it ASAP.

ixjf, 2020-09-05:
As regards inverted colours, that's fixed, and I'll be releasing an updated version of MSIRGB with the fix later today. When it comes to the other issues, I see no reason for that to happen, other than that you may have some other program interfering with the LEDs.

ixjf, 2020-09-06:
Could you try the new version 2.3.0?

FluorescentGreen5, 2020-09-13:
Tried it just now. Colours are no longer inverted and it was able to set the LEDs with no hiccups.

ixjf, 2020-09-13:
Great!

commit d100340f (2020-09-05): Adds support for a bunch of motherboards; fixes #139, #48, #40, hopefully #134 as well. Marks 7B77 and 7B78 as unsupported motherboards as it'sd not guaranteed MSIRGB will work with them.
closed 2020-09-13 by ixjf