ixjf/MSIRGB #142: Works On MSI B450 Pro VDH MAX Perfect With Mystic Light Combined

JohnFlow163, 2020-09-06:
also is It possible where i can find more effect scripts like the heart beat effect?

ixjf, 2020-09-06:
Thanks for the feedback! Are you referring to the heartbeat effect similar to the one in nagisa/msi-rgb? Ill work on it ASAP.

ixjf, 2020-09-06:
I've ported nagisa/msi-rgb's heartbeat effect. However, it doesn't look great on my LED strips. Maybe it's just mine, though. Do you want to give it a try?

```lua
Lighting.BatchBegin()
Lighting.SetBreathingModeEnabled(false)
Lighting.SetFlashingSpeed(0)
Lighting.SetStepDuration(5)
Lighting.SetColour(1, 0xd, 0xd, 0xe)
Lighting.SetColour(2, 0xf, 0xf, 0xf)
Lighting.SetColour(3, 0x9, 0x9, 0xc)
Lighting.SetColour(4, 0xb, 0xb, 0xd)
Lighting.SetColour(5, 0x7, 0x7, 0xa)
Lighting.SetColour(6, 0x8, 0x8, 0xb)
Lighting.SetColour(7, 0x5, 0x5, 0x8)
Lighting.SetColour(8, 0x6, 0x6, 0x9)
Lighting.BatchEnd()
```

JohnFlow163, 2020-09-06:
yes sure il give a try xD

ixjf, 2020-09-07:
How does it look?

JohnFlow163, 2020-09-07:
> How does it look?

tnxxx its actually looking cool, tnxx soo much, hopefully to see more these scripts in future

ixjf, 2020-09-07:
Great!

ixjf, 2020-12-08:
Note: It seems that some motherboards don't act the same way. The heartbeat effect doesn't work well on my motherboard, for example (colours are washed out, loops from colour 1-8 and then 8-1, rather than 1-8 every time, which ruins the effect).

Here's an alternative version which should work for any motherboard:

```lua
-- User-changeable variables
local speed = 130 -- seconds between each iteration of 'colours'
local colours = {
    {0x6, 0x2, 0xa}, 
    {0x6, 0x3, 0xa}, 
    {0x6, 0x4, 0xa}, 
    {0x7, 0x6, 0x9}, 
    {0x7, 0x6, 0x8}, 
    {0x6, 0x6, 0x6}, 
    {0x5, 0x5, 0x5}, 
    {0x3, 0x3, 0x3}}

--
local n = 1

Lighting.SetBreathingModeEnabled(false)
Lighting.SetFlashingSpeed(0)
Lighting.SetStepDuration(511)

while true do
    Lighting.BatchBegin()
    for i = 1, 8 do
        Lighting.SetColour(i, colours[n][1], colours[n][2], colours[n][3])
    end
    Lighting.BatchEnd()
    n = (n % #colours) + 1
    os.sleep(130)
end
```

I ought to figure out why this happens, though. Maybe I'm not configuring some motherboards correctly.

JohnFlow163, 2020-12-22:
> Note: It seems that some motherboards don't act the same way. The heartbeat effect doesn't work well on my motherboard, for example (colours are washed out, loops from colour 1-8 and then 8-1, [...]

thnxx i gave a try and this looks smooth now with bit of purple n white lights

closed 2021-02-04 by ixjf