ixjf/MSIRGB #146: Will this work with a single color led header.

Michaeljaypro, 2020-09-11:


ixjf, 2020-09-11:
Which motherboard and LED strip are we talking about?

Michaeljaypro, 2020-09-11:
I have a B450M Pro M2 Max. It has a JLED1 Header. Usually this was a standard 4 pin RGB header with 12v-G-R-B. But on this and very few select motherboards, its just a 4pin 12v-(no connection)-Signal-(no connection). It allows for single color led strips. When I plug my case fans into it, it does light up, but only one color(red) and i can't change it in Mystic Light. I feel like something in the motherboard is telling my fans to light up red. Is there away around this to allow me to change color? Or do I need to just go buy an rgb controller.

ixjf, 2020-09-11:
If it is a motherboard limitation, then there's really nothing that can be done software-wise. Sorry.

closed 2020-09-11 by ixjf