ixjf/MSIRGB #147: MSIRGB not working

Arsenic333, 2020-09-13:
i followed all the steps but whenever i tyr to launch MSIRGB.exe it says fialed to load driver
[image]
can anyone help me out

ixjf, 2020-09-13:
Do you have Riot Vanguard installed?

Arsenic333, 2020-09-13:
yes

Arsenic333, 2020-09-13:
should i uninstall it
or disable it

Arsenic333, 2020-09-13:
ok it works now thx

Arsenic333, 2020-09-13:
but when i restart my pc it doesnt work
it will only work when i manually disable vanguard first everytime i turn on my pc, Not when vanguard is disabled from start up

ixjf, 2020-09-13:
Unfortunately, Vanguard does not allow MSIRGB to run as it thinks it might be used as a cheat. This is a known issue. In the meantime, you should either stop using Vanguard altogether or disable it when using MSIRGB.

ixjf, 2020-09-13:
I'm closing this as it is a duplicate of #115. Please refer to that issue for status on the fix.

closed 2020-09-13 by ixjf