ixjf/MSIRGB #149: Some colors in 2.3.0 are different than in the previous build

ResoundingSounds, 2020-09-15:
Hello. I have an MSI B450 Tomahawk Max mobo, and I have been using build 2.2.1.2 mostly without issue. I created some scripts to run static colors (with the values reversed to get the proper colors) and everything works perfectly fine. 

But, with the new update I'm noticing that I can't quite get the same colors I could with the previous version. For example, I have a script for a very deep red-orange, and while the values are technically the same (R:0xF G:0x1 B:0x0) vs the reversed (R:0x0 G:0xE B:0xF) for 2.2.1.2, the latter gives me an orange much closer to red, where the one 2.3.0 gives me is closer to yellow.

I would love if the current version could replicate the color from 2.2.1.2. I'm not sure if there is much difference beyond orange being lacking, but I might test out some other colors across versions as well. The main reason I moved away from Mystic Light in favor of this is because I found the color reproduction in MSI-RGB to be much more satisfying.

Thanks for working on this program, I love the new update making this more compatible with my motherboard (aside from this issue)

ixjf, 2020-09-15:
I will look into it ASAP. I've actually noticed this myself with an effect I ported from another project, which used reversed colours.

N30NF15H, 2020-10-27:
I've noticed a similar thing with the B450M Mortar.

2.2.1.2 
Hue wheel - looks good
Police lights - looks good
Pumpkin - colour is blue
Strobe - appears yellow - not sure if that's right

2.3
Hue wheel - colours are different
Police lights - colours are different blue and green
Pumpkin - looks good
Strobe - appears blue - not sure if that's right

ixjf, 2020-10-27:
Thanks. I've been busy because if university so haven't been able to focus on this, but I will when I can.

ixjf, 2021-02-04:
> I've noticed a similar thing with the B450M Mortar.
>
> 2.2.1.2
> Hue wheel - looks good
> Police lights - looks good
> Pumpkin - colour is blue
> Strobe - appears yellow - not sure if that's right
>
> 2.3
> Hue [...]

The behavior for 2.3 is expected. So everything is okay.

ixjf, 2021-02-04:
> Hello. I have an MSI B450 Tomahawk Max mobo, and I have been using build 2.2.1.2 mostly without issue. I created some scripts to run static colors (with the values reversed to get the proper colors) [...]

Perhaps try experimenting with different RGB values? Due to limited accuracy in colours (the chip only supports 4 bits per channel), the colors you get won't always be equal to the one you expect.

ixjf, 2021-08-21:
Any update on this?

N30NF15H, 2021-08-31:
Meant to come back to you sooner on this one. Just tested again.

The colour picker thingy in 2.3 sets the correct colour but 2.2.1.2 appears inverted.

The scripts in 2.3 are all inverted except pumpkin. The scripts in 2.2.1.2 are correct except pumpkin which is blue.

So, if I want to manually set the colours to switch between I use 2.3, if I want to use hue wheel I use 2.2.1.2.

ixjf, 2021-09-03:
Ok, I can confirm the colours are more vivid when inverted. The colours you get with the existing scripts in 2.3 are correct, though. I think I'll add an option to invert the colour channels. That seems to give you a slightly different colour table.

commit c4265b1e (2021-09-03): Fix #149
closed 2021-09-03 by ixjf