ixjf/MSIRGB #150: Only affecting MB Function LEDs on MSI Z370 Gaming Pro Carbon

RageFlow, 2020-09-18:
The MB Function LEDs on the MSI Z370 Gaming Pro Carbon motherboard is a Red colored LED.
So as of now, that board is not "supported".  ;)

ixjf, 2020-09-19:
Are your LED strips connected to the 5050 12V connectors or the WS2812B/CORSAIR connectors?

RageFlow, 2020-09-19:
WS2812B connector and the RGB ligting on the Motherboard itself is not affected by MSIRGB (Except the MB Funtion LEDs)

ixjf, 2020-09-19:
I would have assumed that it would affect the 5050 LED strips, but it's highly unlikely to control the WS2812B connectors, and especially not the CORSAIR one, as these are controlled by separate chips. I'll investigate further about the 5050 connector when I can.

RageFlow, 2020-09-23:
I just went through some log files for Dragon Center (Mystic light), and noticed it was talking about "MB_Model_hex: 7B45"
I don't know if it helps, but if you need me to check some files just hit me up :)

ixjf, 2020-09-23:
I'm aware, but thanks :) Will do.

ixjf, 2021-02-04:
Potentially fixed by #193. If you're willing to try an experimental release, feel free to tell me if it fixes the issue.

ixjf, 2021-08-23:
Hey. So, I took a closer look and this motherboard is actually not supposed to be supported. It was my mistake to list it as supported. Sorry.

closed 2021-08-23 by ixjf