ixjf/MSIRGB #151: Turn off lights at startup script

afsilvasantos, 2020-09-25:
Hi,

Is there any function that allows to disable all lights at startup?

ixjf, 2020-09-25:
You can call `Lighting.SetAllEnabled` with first parameter = `false`. If a script was left running when Windows shutdown, it'll auto-start with Windows as well, so this would have the desired effect.

afsilvasantos, 2020-09-30:
Ok thanks! That worked out. However i run the script, but when i restart the lights are on again. shouldn't the script run automatically now?

ixjf, 2020-09-30:
If you ran the script and didn't stop it at any point, it should have. I see two options:

- You have some other program turning them on.
- This is another case of bug #43.

ronangelo, 2021-07-23:
I have a B450M MB and wanted to disable the lights at startup too. I managed to get it working with this:

```
Lighting.SetAllEnabled(true)

os.sleep( 500 )

Lighting.SetAllEnabled( false )
```

I'm not sure which fixed my issue though. Either setting the `Lighting.SetAllEnabled` to True before disabling OR messing with the service's startup type. I initially set it to Automatic (Delayed Start) but changed it back to Automatic. It's working so far since then.

ixjf, 2021-08-23:
I updated the scripts to fix this issue. You can see the last comment in #159 (the one with the code example) to see how to apply it to your own code.

closed 2021-08-23 by ixjf