ixjf/MSIRGB #154: AMD TRX40

ctrlaltdude, 2020-10-13:
I tried running this on my MSI TRX40 but unfortunately it doesn't work. This surprises me because I would assume that MSI doesn't change their RGB setup with every new mainboard.
I looked through the code but I can't program C++ so I don't understand the workings of this.

Could you please add support for the AMD TRX40 chipset? 

Thank you.

ixjf, 2020-10-13:
> This surprises me because I would assume that MSI doesn't change their RGB setup with every new mainboard.

You would be surprised. Even different revisions of the same motherboard model work differently. For some models, older revisions are not compatible, but newer revisions are.

Can you run this command on the command line so I know which model and revision we are talking about exactly?

`wmic path Win32_BaseBoard get Product, Version`

ctrlaltdude, 2020-10-13:
Really?! That's nuts!

Here's the info:
Product                   Version
TRX40 PRO WIFI (MS-7C60)  1.0

ixjf, 2020-10-13:
Alright, I'll check it out and tell you if it's possible to add support for this motherboard (theoretically, it is, the question is whether it uses the same chip or it would require entire new code to control it).

ctrlaltdude, 2020-10-14:
Cool! Don't put too much effort in it. I don't think the TRX40 chipset is used by many people.

ixjf, 2021-02-04:
Hi. Sorry, I can't add support for this motherboard :(

closed 2021-02-04 by ixjf

ctrlaltdude, 2021-02-04:
No problem.
Thanks for looking into it!