ixjf/MSIRGB #155: Compatibility

PoizenBlood, 2020-10-15:
Was wondering if their was any plans to add support for B550 MAG Tomahawk, as this seems like exactly what I’m looking for, but it is not working at all for me, also thank you for the work put into this.

ixjf, 2020-10-15:
It depends on whether I can reuse the existing code or need new code to control a different chip. I won’t really have time in the coming weeks, though.

PoizenBlood, 2020-10-16:
Completely understandable, thanks for the reply, and the great work.

Fredoias, 2020-10-21:
If possible, when/if you get a chance. I have a B460m Mortar WiFi and was curious if it could be added? Thanks for the work you do.

ixjf, 2021-02-04:
I'm not planning on adding support for new motherboards, sorry. Maybe OpenRGB will do the trick?

closed 2021-02-04 by ixjf