ixjf/MSIRGB #157: Failed to load driver

ShadowKnight2904, 2020-10-21:
[image]

ixjf, 2020-10-21:
Please disable or uninstall Riot Vanguard if you have it installed.

tosyu, 2020-10-23:
Same problem, and I do not have Riot Vanguard installed

ixjf, 2020-10-26:
I have no clue. It's likely another program blocking MSIRGB from loading. Try running Windows in diagnostic mode (Win key + R -> type 'msconfig' -> tick 'Diagnostic mode' -> click 'Ok' -> reboot). Make sure no non-Microsoft services are running then.

tramix32, 2020-11-19:
Faceit antycheat also causes this problem, just uninstall it and run program.

ixjf, 2021-02-04:
This was almost certainly due to an anti-cheat. Please uninstall it. Closing this due to lack of activity.

closed 2021-02-04 by ixjf