ixjf/MSIRGB #159: MSIRGB not running scrips on startup; how do I run them seperately?

maxbear123, 2020-10-27:
Here is my Lua Lighting Script:
``
Lighting.SetColour(1,0,0xf,0xd)
Lighting.SetColour(2,0,0xf,0xd)
Lighting.SetColour(3,0,0xf,0xd)
Lighting.SetColour(4,0,0xf,0xd)
Lighting.SetColour(5,0,0xf,0xd)
Lighting.SetColour(6,0,0xf,0xd)
Lighting.SetColour(7,0,0xf,0xd)
Lighting.SetColour(8,0,0xf,0xd)
``
Every time I wake my computer up or turn it on, my lighting gets reset to red and the script doesn't change it back to blue until I manually open up the application. I even tried setting it up so the app opens on startup, but this didn't fix it either. How can I run the .lua script by myself so that task scheduler runs it on startup?

ixjf, 2020-10-28:
This is a known issue, see #116. Unfortunately, I haven’t had the time nor patience to debug this :(

maxbear123, 2020-10-28:
So there's no way I could make a batch file to run the script myself?

ixjf, 2020-10-29:
No, there isn't, but you can set a timer to delay the execution of the script, or, alternatively, you can run your script in an infinite loop. That gets around the bug.

maxbear123, 2020-10-29:
I figured out another way to solve it. I made the following batch script:
```
@Echo Off
timeout 6
net stop MSIRGB.ScriptService
timeout 1
net start MSIRGB.ScriptService
```
And then made a shortcut to run it as an admin. Then I created a task in task scheduler that ran it whenever I logged on or woke the computer up [(here is how to do that)](https://sumtips.com/how-to/run-program-windows-wakes-up-sleep-hibernate/).
I'll try the delay method too, that sounds more elegant; thanks for the help.

maxbear123, 2020-10-29:
I'm just going to leave this open for anyone else that has the problem.

ixjf, 2020-10-29:
Well, yeah, you can do that too :) Thanks!

ixjf, 2021-02-06:
As I thought, this isn't MSIRGB's fault. MSIRGB runs the script fine and the settings are applied, but _something_ resets the chip afterwards. I have no idea what, but maybe if I can delay the start of the script service, I can avoid these problems.

ixjf, 2021-02-06:
I guess the foolproof way would be for scripts to simply rerun the commands every second or so. Would rather not have to do that, since as little CPU time as it consumes, it's still wasted time, but that's the only foolproof way.

marcxx04, 2021-02-28:
Do I have to deinstall Dragon Center for this to wortk? Im sorry I just found out about this tool a few seconds ago and im not to familiar with coding in general... help :)

closed 2021-08-23 by ixjf

ixjf, 2021-08-23:
Right, so the way I "fixed" this is by adding a Lua class Timer. Basically, just keep resetting lighting settings every X seconds. This is actually good because some other program could change the configuration at any time without MSIRGB knowing it. It's also the way Mystic Light works. For effects that already have an infinite loop (Hue Wheel, Strobe, Police Lights, Heartbeat), this wasn't an issue in the first place and you don't need to change anything. For static effects, you can do:

```lua
local Timer = require('base.timer')

Timer.Set(function()
    -- your Lighting.<something> calls
end, Timer.DefaultInterval)
```

where `Timer.DefaultInterval` is a predefined interval of 3 seconds between updates to the chip (you can use other values, but 3 s seems okay-ish in terms of performance).

It's also a good idea to have ALL Lighting calls within this function passed to Timer.Set, and otherwise for effects with an infinite loop, to have all these calls WITHIN each iteration of the loop, for the reason above.

commit 7dec0963 (2021-08-23): Fix #159, #163, #151, #116, #43 (duplicates)