ixjf/MSIRGB #160: Support for Z490 chipset motherboards

UGHGB, 2020-10-28:
This would be most welcome.
I have MSI Z490 Gaming Plus mobo, CM SickleFlow aRGB fans and Bitfenix Alchemy aRGB strips waiting for software control.

ixjf, 2020-10-28:
I can't guarantee I can add support for that motherboard. Either way, I don't have the time right now to work on MSIRGB, but I'll leave the issue open for when I can. Thanks!

bennodxb, 2020-11-01:
I need z490 too. The msi software is so annoying. Thanks for considering it.

juicebadwik84, 2020-11-23:
Yes please... Would be so awesome if I could use this software with my MPG Z490 Gaming Edge Wifi mobo.  At any point when you have time would be appreciated.

mopperizer, 2020-12-09:
Meg 490 Unify here. Would love to see an update :)

WHCTHBD, 2020-12-14:
Z490-A Pro here. Hoping to see it working without Dragon Center soon!

CZonin, 2021-01-12:
Meg Z490i Unify here, would love to use this!

pawel-zajac, 2021-01-23:
Z490 Gaming edge wifi here. Is it confirmed it won't work with Z490 boards as is, or does it require checking?

ixjf, 2021-01-23:
Likely doesn't, but not sure. I have not gotten any reports of it working anyway.

WoB666, 2021-02-03:
I join with the request for Z490 Gaming edge wifi

ixjf, 2021-02-04:
Sorry, I'm not planning to add support for any new motherboards.

closed 2021-02-04 by ixjf