ixjf/MSIRGB #161: Colours automatically changing between white and desired colour

adamcibulka, 2020-11-01:
Hello, I'm using v2.3.0 and everything was working fine when I hade an RGB strip and two RGB fans connected via one RGB connector to the motherboard. Now I have 4 same RGB fans connected in serial via one RGB connector and whenever I set any colour or mode or just about anything it changes the colours between white (always there) and the colour I have set. I wonder why is that, would anybody know?

**Motherboard:** MSI B360 GAMING PLUS
**System:** W10 64bit v2004

If you need any more info, just ask. I don't know what might you need to analyse this.

adamcibulka, 2020-11-01:
I've also attached here a gif that would represent what the problem is.
It doesn't matter if it is supposed to be a steady colour or some precreated script, it always does this.
For instance the police wheel script goes from white to red to white to blue and so on.
[image]

adamcibulka, 2020-11-01:
I have now tested the OpenRGB and it does exactly the same.
When I was using 2 RGB FANS - everything worked like a charm.
Now it does this.

ixjf, 2020-11-01:
Make sure you don't have Mystic Light or other MSI software installed/running (make sure the MSI_MysticLight service is stopped).

adamcibulka, 2020-11-01:
Okay so I have turned everything regarding MSI Mystic Light off and it still did not work. Then I've tried it again but I've started it as an administrator for a change and now it is working, thank you very much!

Also I really appreciate your work, is there some way I can donate you?

ixjf, 2020-11-01:
> Also I really appreciate your work, is there some way I can donate you?
I don't think that's necessary, but thanks :)

adamcibulka, 2020-11-01:
Okay, you're humble. :) 

Yes, absolutely, you can close it.

adamcibulka, 2020-11-11:
Hello. 
Okay so I've added another fan, the RGB lightning works, but there is the same error again - colour changing from the set colour to white colour with the effect of breathing. I've tried all the things, even the thing that did the trick for me.

Could you maybe look into this a bit more? Because I'm quite desperate and destined to stick with the default colour red.
Also if you need any info from me - just ask.

ixjf, 2020-11-11:
And this only happens when you add another fan? Or Do you think it might have been a coincidence? What happens if you remove that fan?

adamcibulka, 2020-11-12:
No, this was happening before aswell when I unplugged the PC from the source. So basically when I've managed to run it correctly, I had 4 RGB fans. Then the reset of the PC with source unplugged and maybe the CMOS was flashed so it did not remember. So I waited till I've got another fan and now I'm dealing with this. It does the same with OpenRGB, so maybe it is nothing wrong with the software, it's true that Mystic Light just does not work for my motherboard, which is officially supported.

adamcibulka, 2020-11-12:
And also something quite interesting. When I've run it correctly, the yellow colour was indeed yellow. After first normal shutdown and start of the computer, the program would still work, but the yellow would suddenly turn blue and it would stay the same. But that's a minor issue which I've imlemented in my brain. 😄 

Now it is still like that - when I set the yellow colour, it would be blue. But that's fine, the issue is the changing between white and the set colour.

ixjf, 2020-11-12:
It could be MSIRGB failing to setup the LED chip correctly. Do none of the MSI lighting software work?

adamcibulka, 2020-11-12:
Yes. The Dragon Center works correctly regarding fan speed etc., but not the lightning. It works on RGB keyboard for instance, but that is connected via USB. It really bugs me, because it worked flawlessly and now I just can't get it working.

adamcibulka, 2020-11-12:
The Mystic Light software does literally nothing regarding RGB's on motherboard. It is set to red by default but if I turn it off by your software, it also does just nothing.

adamcibulka, 2020-11-12:
And maybe also - when I reset the CMOS, the PC will light up with its red default colour (MSI). So basically everything works, I just think there is some interference with some software or your software fails to setup the LED chip correctly as You've said. Either one it worked before.

ixjf, 2020-11-12:
> Yes. The Dragon Center works correctly regarding fan speed etc., but not the lightning. It works on RGB keyboard for instance, but that is connected via USB. It really bugs me, because it worked flawlessly and now I just can't get it working.

You're saying Dragon Center also doesn't work? But it worked before?

adamcibulka, 2020-11-12:
No no, everything in Dragon Center works except for Mystic Light and that is since the purchase of the PC.

adamcibulka, 2020-11-12:
Basically before the MSIRGB worked, then it did not, when I did reset the CMOS. Then I somehow got it working by turning off the MSI_MysticLight service and maybe something else, run it again and it suddenly worked. And I can't find the winning formula again.

ixjf, 2020-11-13:
What happens if you set a static colour? Does it also change to white?

adamcibulka, 2020-11-13:
Yes, the gif actually show the static colour mode. Meaning I have set every step to the same colour. It changes to white in every mode.

ixjf, 2020-11-13:
It feels more like you have some program running that is messing with the LEDs. If it had been a misconfiguration, a cold boot/clearing the CMOS would have solved it, and it didn't. Could you look through services & startup items and make sure theres nothing weird running?

adamcibulka, 2020-11-13:
Well these are my services, they are sometimes in Czech language, which is my native language, but the names of the programs should be there. Is there anything you would think would interfere with your program? 

And after startup, there is nothing suspicious, but of course there is the Dragon Center.
It could be some combination of shutting down some services, programs, restarting etc., but I can't find it.

[services.txt](https://github.com/ixjf/MSIRGB/files/5538817/services.txt)

Also I'm very very very thankful for your care for this issue.
And of course it might help somebody in the future. 🙂

ixjf, 2020-11-13:
Nothing suspicious at a glance, other than all the MSI Dragon Center AND Mystic Light services running (LightKeeperService, Mystic_Light_Service). Could you uninstall all MSI software and check again?

adamcibulka, 2020-11-14:
Okay so this is what I did and it did not work:

1. Uninstall MSI Dragon Center and MSI SDK
2. Reset CMOS
3. Install VC Redist 2019 x64 again
4. Reboot PC
5. It still does the same

ixjf, 2020-11-14:
Strange. What services are installed and running now?

adamcibulka, 2020-11-14:
Well it's the same as before without the MSI services. Do you think reinstalling Windows might help? 
Because when I clear CMOS, the PC will boot with default red led, which can't be changed with Mystic Light and can be changed with your program. It just alternates between the white colour now.

ixjf, 2020-11-14:
Okay. It stays red until you change the colour for the first time?

adamcibulka, 2020-11-14:
Yes, it stays red by default - which I think means that it is preconfigured in the motherboard. And when I try to change it for the first time, it does this bug. When I try to change it via Mystic Light it does literally nothing, it just stays steady red.

ixjf, 2020-11-14:
I'll check if I missed anything for your specific motherboard as soon as I can, but I can't really tell you when that'll be, since I've got work from university to do.

adamcibulka, 2020-11-14:
Of course, that's understandable. Thank you once again for your care, I really appreciate it. 👍🏻

adamcibulka, 2020-12-12:
Hello again, I just wanted to ask you if you've thought of anything or made any progress in this issue? It would be awesome. 🙂

ixjf, 2020-12-12:
No, I haven't :(

adamcibulka, 2020-12-14:
Okay, I was just checking. 🙂
It just bugs me that the official MSI software doesn't work, yours did, but now it doesn't.
Hopefully it will work again.

adamcibulka, 2020-12-17:
Hello, I've tried the "hot fix" in the other thread, but unfortunately it did not work. 
Hot fix - install Mystic Light and then uninstall it immediately. I even tried some variations and nothing unfortunately. 😞

ixjf, 2020-12-17:
> Yes, the gif actually show the static colour mode. Meaning I have set every step to the same colour. It changes to white in every mode.

I skipped this when you said it. By 'static colour', I meant setting one colour and not changing it again.

adamcibulka, 2020-12-17:
Yes, when I set one colour to every one of the eight squares (for example red), it changes to the white as seen on the gif. Also the colours are somehow reversed - when I set yellow, it changes between white and blue.

ixjf, 2020-12-17:
I'm not sure if you mean you set one colour and leave it alone, or you set one colour but periodically change it. The GIF shows colours changing. I mean to set one colour and never change it again.

Regarding colours inverting, indeed, for model 7B22 (e.g. yours) I have to force inverted colours on the chip. There are quite a few motherboards which are inverted by default, so I have to change this setting accordingly to get the same colour for some RGB value across all motherboards. Maybe I got yours wrong?

Can you try with this DLL? (Replace the one in the MSIRGB folder).
[MSIRGB.DLL.zip](https://github.com/ixjf/MSIRGB/files/5710720/MSIRGB.DLL.zip)

adamcibulka, 2020-12-17:
Okay, it did not help (replacing the DDL).

I'm attaching an image of my settings.
With this settings the colour changes between white and blue (for which I had to set yellow in the boxes).
If I enable **Flashing speed**, it does the same, just with the flashing.
If i enable **Breathing mode**, the effect works, but it is only white colour, no blue.
The white colour is also apparent with your predefined scripts.

[image]

ixjf, 2020-12-17:
Strange. In that case, it doesn't seem to be related to inverting the colours, as I had said. Back to square one.

adamcibulka, 2021-01-05:
Hello again.
Just wanted to ask you if you've progressed anyhow?

And want to wish you best things to happen this year! 🙂

ixjf, 2021-01-05:
Hey. No, not really. I'm really busy with university and haven't been able to really focus on this project. Typically only respond to issues that seem easy to solve =) I haven't given this one much thought, sorry.

adamcibulka, 2021-01-08:
Okay so I've tried it again differently and I will write here precisely all the steps it took me to run **MSIRGB** correctly:

1. Download latest release 
2. Turn off these things, if you have them running:

    - MSI Dragon Center
    - LigthKeeperService (Microsoft Services)
    - MSI Central Service (Microsoft Services)
    - MSI_Companion_Service (Microsoft Services)

3. Run **MSIRGB.exe** as administrator
4. Set your desired RGB lightning

ixjf, 2021-01-08:
And now the problem doesn't occur anymore?

adamcibulka, 2021-01-08:
Yes, now it works and this way I managed to get it working on the first try. 

And when I turned it on, all the lights turned off for 2 - 3 seconds and then it started working.

ixjf, 2021-01-08:
Awesome. So it was interference from MSI programs after all.

adamcibulka, 2021-01-08:
Yes, exactly. But who knows why it works only this way for me. I think the key point was to disable everything MSI related before even turning on your program.

closed 2021-02-04 by ixjf