ixjf/MSIRGB #163: Issue with color palette

nsb-github, 2020-11-28:
In the supported motherboard list, it mentions msi-b350 pro vdh.
My issue is similar to the one mentioned here [https://github.com/ixjf/MSIRGB/issues/113](url)
The lua scripts run fine but the issue is with the color palette.
Let say the color before setting a palette is 'x' (either white or the one set using running scripts).
After applying any color palette say of color 'y', it starts flickering with 'y' on and off back to 'x'

ixjf, 2020-11-28:
I'm assuming you fixed the problem. What was it?

nsb-github, 2020-11-28:
For some reason I thought, there are 8 colors presets and anyone could to be selected at a time as what we have in any photo editing color select table. Later I realised, its always in 5-color transitioning mode. So to keep one color, all 8 must be of same color. This was further clear when I went through the scripts (based on my knowledge of C language).

It was stupid of me to miss that the app clearly states to define 8 colors to switch between.
So, all works fine as I understood the functioning of the app.

Since I got your attention, could you clarify some things...
(1) why number 98 is used to normalize the Hue. All I understand is the Hue resets (to 0) every time 'i' increases to a multiple of 98.
(2) Is there a manual/guide available that lists all the functions that are used or could be used in the lua scripts for further customizations.

ixjf, 2020-11-28:
1) This effect was ported from somewhere else, so I do not know the exact _why_, but it seems to me that it controls how fast you traverse the color palette (the higher the value, the smaller the step). The code should probably be cleaned up :)

2) Smoother transitions aren't really possible because each colour component is stored in a nibble (4 bits). You don't really get 0-255, but 0-16. This is a hardware limitation, nothing I can do :(

3) You can't change intensity of breathing mode either. Also a hardware limitations. It's not possible to change the brightness of the LEDs as far as I could tell.

4) Yes, there is: https://github.com/ixjf/MSIRGB/wiki

nsb-github, 2020-11-29:
Just a thought/guess...
I dont think it would be a hardware limitation as during breathing mode the leds are transitioning very smoothly and do not show "the staircase phenomena" with 16 steps. It seems more likely to be the firmware that is providing limited access to the API.

Anyway, this app is certainly miles better than the Mystic Light.
Thank you very much.

ixjf, 2020-11-29:
By "hardware limitation", I mean that the memory layout exposed by the chip does not allow one to define colours with greater accuracy.

ixjf, 2020-12-09:
Do you need any help or can I close this issue?

nsb-github, 2020-12-10:
I closed the article from my end. It shows " nsb-github closed this 12 days ago".
Is it usual that every once and now, when I start the computer, the LED color profile/script is applied but with a flashing ON and OFF. When I open the app, the script is selected already. I have to manually select something else and reselect the previous one to stop the flashing. This does not happen every time I start the computer but only may be like every 2nd or 4th time.

ixjf, 2020-12-10:
>I closed the article from my end. It shows " nsb-github closed this 12 days ago".

You did, but then you reopened it.

> Is it usual that every once and now, when I start the computer, the LED color profile/script is applied but with a flashing ON and OFF. When I open the app, the script is selected already. I have to [...]

It may or may not be. Can you show me the script? Do you disable flashing inside the script?

ixjf, 2021-08-23:
Hey, you can try the updated scripts (download the 'Scripts' folder individually, please, since I haven't released an update release for MSIRGB yet). Should fix the issue. If not, please reopen this issue.

closed 2021-08-23 by ixjf