ixjf/MSIRGB #165: Vac-AntiCheat compatibility

Ramroll88, 2020-12-04:
Hello I wanted to ask if anyone knows about this software intereferes or is picked up by steams Vac-AntiCheat system?
This concern arises from the fact that "faceit anticheat" and "valorant anticheat" have to be turned on for the program to properly work on the computer. So I am wondering if the program might have some kind of interaction with the Vac-anticheat used to detect thirdparty cheating software being used in the game of counterstrike global offensive?

https://support.steampowered.com/kb_article.php?ref=7849-RADZ-6869

ixjf, 2020-12-04:
From the website, "The VAC system reliably detects cheats using their cheat signatures.". So, it seems it would have to be specifically blacklisted by Steam, unlike Valorant's. I think MSIRGB would be fine, though I can't be certain.

Ramroll88, 2020-12-04:
I understand, its just so frustrating when the only thing I can get from them is:
"Any third party software or other modifications you use in conjunction with Steam should be done at your own discretion. Only cheats (modifications designed to give one player an advantage over other players) will trigger a VAC ban, but we do not provide guidance on what software you can and cannot use alongside a VAC-secured game." "If you have questions about a specific third party application, please contact the developer of the application for assistance."

"How am i supposed to ask the developer if it will cause a Vac ban when no one knows what Vac Net counts as a cheat?

"I'm sorry but we have provided all the available information in our previous message." "Since there's not much more that I can say or do to be of help with this particular issue, I am going to close this help request."

Like telling me yes and no at the same time its so frustrating. I just dont know what to do. 

I even asked if it might detect the files on my pc while im not running the .exe and ban me for it ofc getting the anser "... we have provided all the available information..." 

Purely logically it should absolutely not count as a cheat, the problem is 1) no other software gives me the desired effect expect your beautiful creation 2) i really dont want to get banned and the part about vanguard and faceit anticheat needing to be shut off gives me an illogical (because i have absolutely no knowlage on how this works) conclusion that VacNet might find it bannable? idk...

ixjf, 2020-12-04:
If it considers those anti-cheats as potential cheats, then maybe MSIRGB might be considered as one as well, after all. 

MSIRGB runs a device driver to communicate with the LED chip. This driver is open source and has been signed by someone else, so it would be difficult to alter it (rebuilding a driver requires signing again, and signing is expensive). Still, if there is a bug in the driver, it could give someone control at the kernel level. I guess that's why Riot blocks it.

Do you know if it blocks it only when playing, or always?

Ramroll88, 2020-12-05:
I have not played official valve counterstrike matchmaking since 3 weeks back because of studying but I will have time again soon to do so. 
MSIRGB does not run with Vanguard or Faceit anti cheat running, I have not made any tests with official cs.

(Maybe I misspoke but running faceit anticheat or Vanguard do not count as potenial cheats by Valve Anti Cheat)

ixjf, 2020-12-05:
>MSIRGB does not run with Vanguard or Faceit anti cheat running, I have not made any tests with official cs.

This I know, though I'm not sure there's anything I can do to circumvent it.

ixjf, 2021-02-04:
Yeah, so, I don't think there's any chance to circumvent these AC. Highly unlikely either anti-cheat will allow inpoutx64 to run. Working around the AC is also too much trouble for what it's worth. I'm closing this.

closed 2021-02-04 by ixjf