ixjf/MSIRGB #166: Z270 Gaming Pro Carbon: can only control RAM status LEDs, eventually starts crashing any time a change is applied

eduardog3000, 2020-12-06:
MSIRGB has no control over most of the lights on my Z270 Gaming Pro Carbon.

The only lights it does control are the RAM status lights, which MSI Gaming APP calls "MB Function LED". These lights seem to just be red, as even the official app can't control the color. So MSIRGB basically just can control whether they are on or not.

In addition, after messing around with MSIRGB for a bit, it started crashing every time I tried to apply a change, leaving it stuck on what it was before the first crash ("police lights" making the RAM lights blink really fast... yay). This happens for the regular "Apply" button, as well as "Disable All Lighting", "Stop Running Scripts", and trying to run a different script. Manually killing `MSIRGB.ScriptService.exe` fixed this.
 
In short, it has greater than 0 control, but not much, and it's a little finicky.

ixjf, 2020-12-06:
This motherboard is not currently supported, so it makes sense. I can see if it might eventually be possible to support it, but given the behaviour described, perhaps not (without major changes to my code).

deekstercaddy, 2021-01-20:
Hi, new to github and this software. I have a Z270i Gaming Pro Carbon AC motherboard (Mini-ITX form factor) and would love to have control over the motherboard's RGB output. Is there anything I can offer to give you more information about it? (assuming you aren't able to obtain one for testing etc)
My ideal purpose is to change the color based on CPU temp or CPU fan RPM, MSI's options are horrible and I'm willing to contribute however I can.

ixjf, 2021-01-20:
Hm, maybe. I'll look into this around mid February.

ixjf, 2021-02-04:
I will not be supporting any extra motherboards, sorry. As for the Z370 GAMING PRO CARBON AC, it _should_ be compatible, though I have missed some initialization steps, which I'm working on. Closing this issue.

closed 2021-02-04 by ixjf