ixjf/MSIRGB #167: New script

ImJuampi, 2020-12-11:
Hello, I wanted to know if you could help me with a script that I wanted to make, what I would like to do is a transition with the colors of my preference, similar to the pumpkin script but with other colors.
I'm sorry if my english is bad.

ixjf, 2020-12-11:
What have you tried?

ImJuampi, 2020-12-11:
So far I have not done much since I do not know much about programming, what I try is to modify the pumpkin script colors to see if it works with that.
I wanted to know if you could help me by telling me what commands I should use.

ixjf, 2020-12-11:
There is a wiki with all the info you need: https://github.com/ixjf/MSIRGB/wiki

You can also try looking into the Heartbeat effect (make sure it's the same you find here on GitHub, I updated it one or two days ago). Should give you an idea of how to create those transitions.

ImJuampi, 2020-12-11:
ok, I'm going to try, since what I want is that it does not change instantly to another color, what I want is that it gradually changes between the selected colors.
Thanks for the help

ixjf, 2020-12-11:
That is a bit more complicated unless you want simple linear interpolation between set colours. You can try googling ready-made algorithms for this (there should be tons on Stack Overflow, just google it) and translating to Lua.

ImJuampi, 2020-12-12:
I've been searching and found this:
https://stackoverflow.com/questions/25668828/how-to-create-colour-gradient-in-python
https://stackoverflow.com/questions/59181825/lua-script-for-rgb-control
https://stackoverflow.com/questions/63201126/lua-script-to-control-pc-rgb-lighting
https://stackoverflow.com/questions/5670372/looking-for-help-in-c-sharp-with-rgb-color-cross-fade
https://stackoverflow.com/questions/13488957/interpolate-from-one-color-to-another
I don't know if this works, but I know practically nothing about programming, so if someone wants to do it, I'll leave the links to the information I found in case someone wants to do something with it.

Pd:
Adding to this I leave a script (based on heartbeat.lua) with which you can place the color you want and this when starting the pc will be activated (to deactivate the breathing mode go to Lighting.SetBreathingModeEnabled (true) and change the value (true) to (false))
[Breathing red.txt](https://github.com/ixjf/MSIRGB/files/5682148/Breathing.red.txt)

ixjf, 2020-12-12:
if your effect is supposed to use one single colour, then Heartbeat is overkill and inefficient (it resets the colour every 130ms). Pumpkin would be ideal. Heartbeat only makes sense if you plan on adding more colours.

ImJuampi, 2020-12-12:
It does not change color since you only leave one colour, try it and you will see that the lights will not change and it will be only one color

ixjf, 2020-12-12:
That is what I said. The problem is that it is inefficient.

closed 2021-02-04 by ixjf