ixjf/MSIRGB #170: Your MB not on the list, then failed to load drivers.

mbalci420, 2020-12-15:
Hi. I got msi b450-pro-m2-max. I am trying to load the program and following all the instructions exactly but, when I start msirgb.exe it says your motherboard not on the list, and then I click to continue, it fails again and warns me that failed to load drivers. What can I do ? Help pls.
[image]
[image]
[image]
[image]

ixjf, 2020-12-15:
Do you have any anticheat installed? Vanguard, FACEIT, etc.

mbalci420, 2020-12-15:
Nope. I don't have any of them.

mbalci420, 2020-12-19:
Can anyone please help me? :(

ixjf, 2020-12-20:
https://github.com/ixjf/MSIRGB/issues/132#issuecomment-674405037

Can you try this debug build and tell me the output of the debug log? (Explained there how to do that)

mbalci420, 2020-12-20:
The same thing happend. 
Debug_log :
start_drv: StartService failed with error 647
load_drv: start_drv() != ERROR_SUCCESS
load_drv failed

ixjf, 2020-12-31:
This sort of error is usually caused by some program interfering with MSIRGB's driver. Can you try running MSIRGB in a sort of 'safe mode' boot? Just disable all non-Windows services and startup items (from 'msconfig' and Task Manager) and reboot, then try.

mbalci420, 2020-12-31:
Nope :( Sorry but the same thing happened again. I'm done with it and sick of this trash motherboard. Thanks for your struggle. Happy new year.

ixjf, 2020-12-31:
I guarantee that's not a problem with your motherboard, though it is possible MSIRGB won't support your motherboard either way. If you have any games installed, I suggest you check if they don't include any anti cheat.

Happy New Year.

ixjf, 2021-02-04:
Did you fix this?

mbalci420-hub, 2021-02-04:
Nope I didn't

ixjf, 2021-02-04:
Can you open a command prompt and run these commands?

```sc query type=driver > drivers.txt
tasklist > userprc.txt```

It should create the files 'drivers.txt' and 'userprc.txt' in the directory where you ran the command. Please upload these.

mbalci420-hub, 2021-02-04:
I'm sorry but it's still the same shit again and again. :( I really appreciate your effort but as I said before, this motherboard sucks.

ixjf, 2021-02-04:
These commands are not meant to fix your issue, but to help me figure out what is causing it.

ixjf, 2021-02-05:
Will you try this, or should I close the issue?

ixjf, 2021-02-22:
I'm closing the issue then.

closed 2021-02-22 by ixjf