ixjf/MSIRGB #171: Second led has a different color - Hua Wheel script

ImFilippo, 2020-12-17:
Hi, I downloaded your program and I must say that it works really well; however I have a small problem, as you can see for yourself from the video under, the second led of the motherboard it has a different color from the others if I use the Hua Weel script or any other script.
Is it my problem? Is there any way to fix it?
Thanks!

My PC:
- Motherboard: B450 TOMAHAWK MAX
- OS: Windows 10 20H2 (latest build)

[https://streamable.com/h3xiew](url)

ixjf, 2020-12-17:
Very strange. MSIRGB can't really set per-LED colours, so it's either a motherboard error or some motherboard-specific thing that MSIRGB is configuring wrong. Does this happen when using, say, Mystic Light?

ImFilippo, 2020-12-17:
> Very strange. MSIRGB can't really set per-LED colours, so it's either a motherboard error or some motherboard-specific thing that MSIRGB is configuring wrong. Does this happen when using, say, Mystic Light?

There is no problem with mystic light , only with Msirgb. I don't know what can I do...

ixjf, 2020-12-18:
Nothing. I’ll look into it as soon as I can, but I can’t promise anything as I’m busy with university.

ixjf, 2021-02-04:
Does this still happen? It's weird that no one else has reported this issue on this motherboard. Are you sure this doesn't happen with Mystic Light?

ImFilippo, 2021-02-04:
> Does this still happen? It's weird that no one else has reported this issue on this motherboard. Are you sure this doesn't happen with Mystic Light?

Sorry if I didn't let you know anything more, over time it will seem strange but now the LEDs work well and the color is coordinated for every LEDs.
I have not installed mystic light nor your program, I really don't know what the problem could have been ...

ixjf, 2021-02-04:
Weird. Well, I'll close this issue for now then. If you have any questions, just reply here or open a new issue.

closed 2021-02-04 by ixjf