ixjf/MSIRGB #173: Adding support for B550 MBs (specificaly B550 Tomahawk)

arjudedjet, 2020-12-29:
Hello, 

would it be possible to add support for "new" B550 motherboards such as Tomahawk? MSI Dragon Center is terrible piece of software, so any other alternative would be appreciated. 

Thanks for any answer and a have a great day.

arjudedjet

ixjf, 2020-12-31:
Maybe. I don't currently have time to do that sort of thing. I would only do it if I could reuse the existing codebase. If the motherboard uses a different chip (which many do) than the ones supported by MSIRGB, then it would be impossible for me to test it before releasing.

ixjf, 2021-02-04:
Right, so, as I said, these new motherboards seem to use a different mechanism for controlling LEDs, so I won't be adding support for them. You can look into OpenRGB, see if it works on your motherboard.

closed 2021-02-04 by ixjf