ixjf/MSIRGB #174: [Q] MSIRGB settings saved on motherboard? any way to go back to default motherbard settings?

wkewinw, 2020-12-30:
Hey :) for me when i reinstall windows it seems that the MSIRGB settings stick like it is saved on the motherboard. I have the RGB turned off currently. At first I thought it wouldn't stick if I Reinstalled Windows but that's good thing!

but is there a way to go back to default like I never used MSIRGB? just wondering.
also wondering if the MSIRGB settings stuck for you if/when you Reinstalled windows as it did for me :)

ixjf, 2020-12-30:
Static settings are saved on the chip's memory, so yes, it has nothing to do with Windows. You'll have to pull the CMOS battery out for a while in order to reset it (or turn off the power supply and press the power button for about 10 seconds).

wkewinw, 2020-12-30:
ah i see :) is it possible even a bios flash would work? that seems to reset everything in bios. i haven't tried that yet tho. but i will soon when they release a new bios update.

"(or turn off the power supply and press the power button for about 10 seconds)" do you mean turn off power supply and press power button for 10 sec while the power supply is still turned off :O ?

ixjf, 2020-12-30:
> ah i see :) is it possible even a bios flash would work? that seems to reset everything in bios. i haven't tried that yet tho. but i will soon when they release a new bios update.
Unlikely, but you could try.

> 
> "(or turn off the power supply and press the power button for about 10 seconds)" do you mean turn off power supply and press power button for 10 sec while the power supply is still turned off :O ?
Yes. By pressing the power button while no external power is on, the capacitors will discharge. I'm not sure this will work with all motherboards. It works with mine. But removing the CMOS battery will reset the settings, that's guaranteed.

wkewinw, 2020-12-30:
okay thanks :D !

wkewinw, 2020-12-31:
> 
> Yes. By pressing the power button while no external power is on, the capacitors will discharge. I'm not sure this will work with all motherboards. It works with mine. But removing the CMOS battery will reset the settings, that's guaranteed.

what about if i installed MSI Mystic Light? would that make it go back to default? i have to take out my GPU if i want to access my cmos battery hehe

wkewinw, 2020-12-31:
oh right i could try that power button thing also

ixjf, 2020-12-31:
> what about if i installed MSI Mystic Light? would that make it go back to default? i have to take out my GPU if i want to access my cmos battery hehe

Usually, yes, that will work too.

wkewinw, 2020-12-31:
ended up taking out the GPU anyway and took out the battery for a while hehe xD thanks :) i read that msi had risks of bricking motherboards :O or maybe that's just newer models?

ixjf, 2020-12-31:
What do you mean? MSI?

Also, why did you feel it was necessary to go back to default settings?

wkewinw, 2020-12-31:
yeah msi motherboards apparently has sensitive rgb chips. like more sensitive than other manufacturers but idk that's just what i read.

eh.. i sometimes get worried and anxiety about "clean-ness" in systems hehe but that's just my problem hehe xD ... it has nothing at all to do with your software ur software is great! very in my experience it did exactly what i wanted. then i just happened to read online "msi motherboards rgb chips thing is very sensitive and high risk for bricking" and it gave me anxiety.

at the same time i know that it was probably unnecessary to go back to default settings but the anxiety man.. for example, i can reinstall windows 8 times in a row until it feels right/good

i didn't mean to get to personal but everything is fine now i feel good ^^

wkewinw, 2020-12-31:
but just to be clear.. MSIRGB does NOTHING to the windows operating system? is that correct? it only changes the bios?

ixjf, 2020-12-31:
I've never heard of that and there's absolutely nothing to be afraid of if your motherboard is supported.

>but just to be clear.. MSIRGB does NOTHING to the windows operating system? is that correct? it only changes the bios?

It doesn't change the BIOS. LED config is stored in a special chip, hence why you need to remove the CMOS battery and why flashing the BIOS likely does nothing to restore the default settings. MSIRGB uses an open source driver (inpoutx64) to "talk" to this chip, and it installs and runs a service to keep scripts running in the background, but everything is cleaned up *every time* you toggle off scripts, so you can simply delete all files afterwards and it'll be as if MSIRGB never ran.

wkewinw, 2020-12-31:
I see :) I never used any of the scripts. I only used it to turn of the rgb light. And tried changing static color once. Very nice little software actually. I will probably end up using it in the future again. 
Is it possible to add a reset default settings in the software? Or does that not work and can only be reset by the methods u mentioned?
I have the MSI B450M Mortar Max. With no issues :)

ixjf, 2020-12-31:
The only "reset" I could do was figure out the default settings for all motherboards and apply them in the same way I apply any other setting. But it's completely unnecessary.

closed 2020-12-31 by wkewinw