ixjf/MSIRGB #178: MSI MS-B090

Pepaco, 2021-01-04:
This motherboard is not detected as MSI, any chance to include it?
Thanks!

ixjf, 2021-01-04:
Can you run this command on a command prompt and tell me the result?

`wmic path Win32_BaseBoard get Product`

Either way, this motherboard isn't in the list of supported motherboards, so MSIRGB _might_ not work.

Pepaco, 2021-01-04:
Sure, no problem!

`Product`
`MS-B090`

ixjf, 2021-01-04:
[MSIRGB.DLL.zip](https://github.com/ixjf/MSIRGB/files/5765316/MSIRGB.DLL.zip)

I updated the code to let it run on your motherboard. **This does NOT mean MSIRGB will work on your motherboard, and I cannot guarantee nothing bad will happen if you run it, so do it at your own risk**.

Pepaco, 2021-01-04:
Hi, thanks @ixjf 
But I've replaced the file and still getting the same "motherboard's vendor was not detected to be MSI". Is there anything else I need to do with this file?
Thanks again!

ixjf, 2021-01-04:
Ah, yes. I'm an idiot. MSIRGB checks the Manufacturer label, not Product. Tell me the result of running this command instead:

`wmic path Win32_BaseBoard get Manufacturer`

Pepaco, 2021-01-04:
Not problem at all! Happens :P

`Manufacturer`
`MICRO-STAR INTERNATIONAL CO., LTD`

ixjf, 2021-01-04:
[MSIRGB.DLL.zip](https://github.com/ixjf/MSIRGB/files/5765378/MSIRGB.DLL.zip)

Here. Should work now.

Pepaco, 2021-01-04:
Now it does, thank you so much!

ixjf, 2021-01-04:
So, does it work properly? Or does it just run?

Pepaco, 2021-01-04:
Just run, looks like nothing happens with the lights. But I'm not able to get it work with this or MSI Mystic Light, or any kind of MSI Center, so maybe it's a problem with my computer or anything else.
But thank you again for your help!

ixjf, 2021-01-04:
Yeah, I couldn't really find any info on that motherboard either. Looks like some motherboard specifically designed for a pre-built PC? Or did you buy it separately?

Pepaco, 2021-01-04:
Yes, I think maybe it's a specifically designed, also has a few years now 😅
https://www.msi.com/Desktop/Nightblade-MI2/Specification

ixjf, 2021-01-04:
It doesn't seem to have any RGB LED connectors on the motherboard? If that's the case, then I'd say it's very unlikely to have support for controlling LEDs.

Pepaco, 2021-01-04:
But it has! A few years ago I was able to control them by some kind of MSI Software (Gaming Center I think), but now I'm not able to turn them off with any kind of software... As I say, maybe some kind of conflict or incompatibility... Who knows :S

ixjf, 2021-01-04:
Hmm, don't really know. The last versions of Mystic Light that I've looked into don't seem to have any code for MS-Bxxx motherboards, indeed, so it's difficult for me to do anything about it.

Pepaco, 2021-01-04:
Don't worry, I'll continue investiganting this, I'll let you know in case I find something. Thank you so much again :)

ixjf, 2021-02-04:
I'm closing this since I don't think I can support this motherboard. If you do find something useful, though, feel free to share.

closed 2021-02-04 by ixjf