ixjf/MSIRGB #181: Uninstaller?

Lithopop, 2021-01-14:
So I decided to be one of the brave people who tried it on an untested motherboard (MSI b550m mortar) and it was incompatible.
I've reflashed my bios, but I'm assuming all the issues that I'm having right now are windows related.

I've since deleted MSIRGB, and reinstalled Mystic Light when hangs when I hit apply and goes black, but after rebooting the color changes stick.
When the computer comes on, the fans are all orange (Pre-MSIRGB installation color), but change when booting into desktop. Wraith Prism Cooler program still functions just as it always had. 

Thoughts on cleaning this up? I'm guessing that some DLL somewhere didn't get removed (or written over) when everything was reinstalled.

ixjf, 2021-01-14:
Reflashing the BIOS won't restore the default settings. The only way is to clear the CMOS chip by unplugging the computer from the power outlet and removing the CMOS battery on the motherboard for ~10 minutes. Turning off the power supply and pressing the computer case's power button also does the trick with my motherboard.

If you've deleted MSIRGB, there's little chance anything was left behind. All the files from MSIRGB are the ones that you download. No others stay in the system when you turn off the scripts. And if you hadn't turned off the scripts, you wouldn't have been able to delete those files in the first place.

ixjf, 2021-02-04:
I'm closing this since I already answered your questions. If you need further help, add a comment to this issue.

closed 2021-02-04 by ixjf