ixjf/MSIRGB #192: Slowing down Hue Wheele

Ramroll88, 2021-02-03:
Is there any way to slow down the cycling of each color in the Hue Wheele mode, I LOVE IT but would wish to have it cycle slower. 

Thanks in advance / 227

ixjf, 2021-02-03:
Yes. You can modify the script. See: https://github.com/ixjf/MSIRGB/blob/master/Scripts/Hue%20Wheel.lua

This file will be in your Scripts/ folder.

Changing `colour_step` will make transitions between colors slower.

ixjf, 2021-02-04:
I think this is a problem with the chip, not MSIRGB. The chip only supports 4 bits per channel. For example, RGB (F, 1, 0) and RGB (F, 0, 0) are completely different. The former is pumpkin orange, the latter is the real red you want - and all that changed was one channel by one unit. You can't choose colours with any higher precision.

ixjf, 2021-02-04:
By the way, I fixed the Hue Wheel script, it wasn't using the colour_step variable as it should. You should download the new file (from the link above).

Ramroll88, 2021-02-04:
Okey i understand, thanks for your great work as always and your active answering <3 you are appreciated :)

closed 2021-02-04 by ixjf