ixjf/MSIRGB #193: Z370 GAMING PRO CARBON AC not compatible, but should be

ixjf, 2021-02-04:
@ixjf I can confirm the same behavior on the Z370 GAMING PRO CARBON AC, even though the Z370 are listed as "should work". Any chance we can get these MBs to work?

_Originally posted by @Eyekatcher in https://github.com/ixjf/MSIRGB/issues/166#issuecomment-763968614_

I missed some initialization steps for this motherboard. Working on it.

ixjf, 2021-02-04:
[MSIRGB.DLL.zip](https://github.com/ixjf/MSIRGB/files/5927881/MSIRGB.DLL.zip)

Could someone provide feedback on whether this fixes the issue for the Z370?

Eyekatcher, 2021-02-04:
Will test this in approx. 2-4 hours, will give you feedback as soon as I can.

Eyekatcher, 2021-02-06:
Still the same issue, only the function LEDs are affected by changes in the software. Used 2.3.0 with the DLL you attached above.

ixjf, 2021-02-06:
Ok, will keep digging.

Arakon, 2021-03-11:
Just to confirm (and track this report), same board, same issue here.

ixjf, 2021-03-11:
Thanks. I got busy with university again so it will take a bit :( Sorry.

ixjf, 2021-08-23:
Seems like I may have been wrong and this motherboard is actually not "supposed" to be supported. Sorry :(