ixjf/MSIRGB #203: GUI doesn't seem to be updated?

FeSong, 2021-02-22:
Just saw the new update about apply to all colours, really waited a long while for this as it makes not having to slide it each time far better. However, after following install instructions it doesn't seem to update or get the latest? The GUI is still the same as before without the Apply to all colours bit?

FeSong, 2021-02-23:
Ah after reading more I guess the new version has not yet been released?

ixjf, 2021-02-23:
That's right. I can send you a build with the latest changes, though.

[Release.zip](https://github.com/ixjf/MSIRGB/files/6030762/Release.zip)

FeSong, 2021-02-24:
> That's right. I can send you a build with the latest changes, though.
> 
> [Release.zip](https://github.com/ixjf/MSIRGB/files/6030762/Release.zip)

Thank you very much man, you've created something amazing here! Wish you the best and thank you again :)

closed 2021-02-24 by FeSong