ixjf/MSIRGB #208: MSI B450 Gaming Pro Carbon AC compatibility

d4rkne55, 2021-03-10:
So, I was searching for an alternative to the bloaty MSI software and checked the support list and saw the MSI B450 Tomahawk listed and supported, but didn't see the Gaming Pro Carbon. I was curious; to my knowledge those boards are very similar, just that the Gaming Pro Carbon has a WiFi module that the Tomahawk doesn't and maybe some other minor differences, so I was wondering why just the Tomahawk is listed and the Gaming Pro Carbon is missing completely.

I wanted to try the software, but it directly showed me a warning before starting, including a mention of "irreversible damage".
Could you tell me what could go wrong, generally and maybe specifically for that situation? I mean, writing bad data to the chip could generally make it malfunction, but could that be restored somehow? And what about this MB; no issue expected here? I would be willing to help out.

Also, checking the code a little bit I found mentions of "NCT controllers" - after some research I found out this is a series of Super I/O chips from Nuvoton. So I checked what I have on my MB and it also is one - the **NCT6797D**(-M).
I then researched what chip the Tomahawk (Max) uses and according to [KitGuru](https://www.kitguru.net/components/motherboard/ryan-martin/msi-b450-tomahawk-motherboard-review/3/) and [this image](https://www.hardware-journal.de/images/Bilder/2019/Test/MSI-B450-Tomahawk-Max/MSI-B450-Tomahawk-Max-4.jpg) it uses exactly the same one.
So I am still wondering what could be the issue? Shouldn't it just work or need minor adjustments, like for different number of zones? I checked #103 and don't see why the input of _XenSide_ is taken for granted, being confirmed and that's it.

ixjf, 2021-03-10:
> So, I was searching for an alternative to the bloaty MSI software and checked the support list and saw the MSI B450 Tomahawk listed and supported, but didn't see the Gaming Pro Carbon. I was [...]

You can't tell whether a motherboard will be supported or not by whether they seem similar in functionality. Different revisions of the same motherboard model can be different when it comes to LED control.

>I wanted to try the software, but it directly showed me a warning before starting, including a mention of "irreversible damage".
Could you tell me what could go wrong, generally and maybe specifically for that situation? I mean, writing bad data to the chip could generally make it malfunction, but could that be restored somehow? And what about this MB; no issue expected here? I would be willing to help out.

The warning message is there for all motherboards for which I cannot assure MSIRGB will work. As you can see in the list of working motherboards, some are reported to work but I say that a warning message will show up anyway - this is because I couldn't confirm for myself, by own research, that these motherboards work 100% the same way, but someone else reported them to be so). For all the others, I guarantee that MSIRGB _should_ be compatible, and if it isn't, it's 100% due to a bug.

Most likely any damage caused to any of these motherboards would be of the changing-wrong-config kind and could easily be restored, yes. But on a motherboard that is not on the list, it's completely undefined behavior. The chip might not be on port 0x4E, and I could be accidentally accessing the wrong chip, or god knows what. Basically, I don't know the extent of the damage possible, so I prefer to be on the safe side - if you run MSIRGB on a motherboard I don't specifically say it is supported, then you're completely on your own.

>Also, checking the code a little bit I found mentions of "NCT controllers" - after some research I found out this is a series of Super I/O chips from Nuvoton. So I checked what I have on my MB and it also is one - the NCT6797D(-M).
I then researched what chip the Tomahawk (Max) uses and according to KitGuru and this image it uses exactly the same one.
So I am still wondering what could be the issue? Shouldn't it just work or need minor adjustments, like for different number of zones? I checked #103 and don't see why the input of XenSide is taken for granted, being confirmed and that's it.

The model of the chip is not really relevant. All of the supported (and many unsupported) use NCT6795/6797 chips, but they're programmed to do different things, as far as I can tell. It _could_ be that it could work with minor adjustments, but I haven't found reliable info on how it's supposed to work. Or it could be that it would require adding whole new logic to control different chips. 

In this particular case, the reason why it was "taken for granted" (it wasn't) is that at that point I had done some research and concluded that it, indeed, wasn't compatible.

In general, unless MSIRGB doesn't work with a motherboard that it should support, I won't be adding support for more motherboards - this requires quite a bit of research and testing, which requires time and access to these motherboards, neither of which I have right now.

d4rkne55, 2021-03-10:
> You can't tell whether a motherboard will be supported or not by whether they seem similar in functionality. Different revisions of the same motherboard model can be different when it comes to LED control.
> [...]
> The model of the chip is not really relevant. All of the supported (and many unsupported) motherboards use NCT6795/6797 chips, but they're programmed to do different things, as far as I can tell. It _could_ be that it could work with minor adjustments, but I haven't found reliable info on how it's supposed to work. Or it could be that it would require adding whole new logic to control different chips.

I see, didn't know. Thought it would be very similar if it has the same chip, and just different chips being an issue to implement.

> Most likely any damage caused to any of these motherboards would be of the changing-wrong-config kind and could easily be restored, yes. But on a motherboard that is not on the list, it's completely undefined behavior.

I see, so you can't tell if it should be safe to try or not on my MB, including the situation of a very similar board with the same chip working.

> The chip might not be on port 0x4E, and I could be accidentally accessing the wrong chip, or God knows what.

For the same chip also?

> In this particular case, the reason why it was "taken for granted" (it wasn't) is that at that point I had done some research and concluded that it, indeed, wasn't compatible.

I didn't mean this to be rude btw, so sorry if it came off like that. That was just how I saw it, without more context - some guy that doesn't seem affiliated with the project commented that it's not compatible and you commented "Indeed", without an obvious reason for the confirmation.^^
So, it uses the same chip but does control the LEDs in a pretty different way to the Tomahawk and there's (currently) not enough information about it, that's how I understand it now.

Also, thanks for the rather extensive answer. :)

ixjf, 2021-03-11:
>I see, so you can't tell if it should be safe to try or not on my MB, including the situation of a very similar board with the same chip working.

In your particular case, someone had commented that they tried it and nothing bad seemed to have happened, MSIRGB simply didn't do anything. But it could have done something bad and they didn't notice. Really can't tell.

>For the same chip also?

Perhaps not, but I really can't guarantee it.

>I didn't mean this to be rude btw, so sorry if it came off like that. That was just how I saw it, without more context - some guy that doesn't seem affiliated with the project commented that it's not compatible and you commented "Indeed", without an obvious reason for the confirmation.^^

It didn't come across as rude, don't worry.

>So, it uses the same chip but does control the LEDs in a pretty different way to the Tomahawk and there's (currently) not enough information about it, that's how I understand it now.

Heh, more like I don't really want to spend much time on MSIRGB, and I don't want to make someone else my guinea pig (risk + slow development). I could find the info if I needed it.

closed 2021-08-20 by ixjf