ixjf/MSIRGB #212: Performance drop after running MSIRGB on B-350

moyds4, 2021-03-26:
LED's did turn off and program worked but now my computer is performing significantly worse. I deleted program and even re-flashed bios and games are still launching twice as slow and I'm getting half as many frames. Not looking to blame anyone or anything because there is a chance that this is because of something else I did but if anyone knows a fix for this please let me know.

ixjf, 2021-03-26:
You deleted _all_ the files? Including `MSIRGB.ScriptService.exe`? What are your current CPU, GPU temperatures and fan speeds?

moyds4, 2021-03-26:
I did but I figured out the issue I believe, I restarted my computer when its performance declined and accidently installed a windows update in the process. I am going to wipe and reformat my C drive, which I also installed MSIRGB on and then I will update here if it is preforming like it was before. To answer your question fans and temps have been same before and after.

ixjf, 2021-03-26:
That seems like a drastic measure to take, but you do you.

ixjf, 2021-03-31:
How did it go?

ixjf, 2021-08-20:
Closing this due to lack of info and activity from author.

closed 2021-08-20 by ixjf