ixjf/MSIRGB #213: Unable to run

EagL3-GIT, 2021-04-03:
[image]

ixjf, 2021-04-03:
Vanguard and other anti-cheats must be disabled before running MSIRGB.

Gash3d, 2021-05-15:
I am getting this same issue however i am not running a single anti-cheat. I noticed that i was getting this same type of problem when trying to use MSI Center/Dragon Center. My motherboard is the x470 Gaming Plus.

ixjf, 2021-05-16:
Whatever problem you were having with MSI programs is not the same one as this since MSIRGB is completely independent from these other programs.

Either way, past experience tells me that you probably **are** running an anti-cheat, though unaware of it. Do you have any games installed?

Gash3d, 2021-05-16:
Games i have installed are these paired with League of Legends & Valorant. I have Riot Vanguard disabled at startup and have restarted my PC many times afterwards. 
[image]

ixjf, 2021-05-16:
It seems that (at least) Apex Legends also uses a kernel-level anti-cheat that could be causing these problems. Could you run the following command on a command prompt and upload here the output file?

`driverquery /fo table /v > instdrivers.txt`

This will create a file `instdrivers.txt` in the directory where the command prompt was started. It will contain a list of all installed drivers on your system.

Gash3d, 2021-05-16:
Here's the file 
[instdrivers.txt](https://github.com/ixjf/MSIRGB/files/6489853/instdrivers.txt)

ixjf, 2021-05-16:
It seems that the Riot Vanguard driver is still running on your system.

Gash3d, 2021-05-16:
I had no idea. I thought Stopping it at startup was enough. I uninstalled it and the software works now. Thank you so much!!!

ixjf, 2021-05-16:
I believe there is a way to disable Vanguard without uninstalling it, but I'm not sure. I can't tell you precisely how because I've never used it.

closed 2021-05-16 by ixjf