ixjf/MSIRGB #242: Motherboard vendor not detected to be MSI

elciebell, 2021-08-28:
I have an MSI MEG Z490 Godlike motherboard. Your software doesn't recognize it.

[image]

ixjf, 2021-08-28:
Likely not supported either, but we'll see. Please run the following command on a command prompt and comment here the output:

`wmic path Win32_BaseBoard get Manufacturer`

elciebell, 2021-08-29:
Manufacturer 
Acidanthera

I'm running Windows on a separate SSD on a Hackintosh machine. macOS 11.5.2, Win 10 Pro, Zorin OS 16 Pro all on separate SSDs. My bootloader is Opencore 0.7.2, which is where the Acidanthera is coming from.

ixjf, 2021-08-29:
I'm not going to bother making MSIRGB somehow recognize that you're running a Hackintosh, if that's even possible. OpenCore overrides information about your motherboard. Is there some way you can boot Windows through Windows' own bootloader, without going through OpenCore?

elciebell, 2021-08-29:
I’ll try and let you know.

ixjf, 2021-09-02:
Did you solve it?

elciebell, 2021-09-02:
There is no way for me to install windows separately from my other OSs, unless I wipe all the disks clean and do a fresh windows-only install. Something I’m not prepared to do. I’ll put up with not being able to control my RGB devices.

Stephen

ixjf, 2021-09-02:
Can't you just boot Windows without going through OpenCore? Or does OpenCore modify the Windows bootloader files?